WeActStudio/BluePill-Plus #4: Does the GD32F303CCT6 have a FPU?

djix123, 2021-03-15:
When I try and compile code using hardware floating point I get a HardFault exception on the first instruction that tries it execute a floating point instruction. The fault indicated is NOCP despite have written to 0b11 to CP_10 and CP_11 of the CPACR.

I've tried with IAR, Keil and GCC each with the same results. Indeed, if I take the example 04-MSC_internal_flash and then adding some floating point arithmetic to main() results in the HardFault. This can be cured by changing the 'Floating Point Hardware' from 'Use Single Precision' to 'Not Used'. The compiler no longer generates FPU instructions and the code works.

WeActTC, 2021-03-16:
Yes, We tested that its FPU was OK,
Keil 'Floating Point Hardware' set 'Use Single Precision'
```
float a=0;
a=1.f/3.f;
```
Result is `a=0.333333343`

djix123, 2021-03-16:
That's interesting it doesn't work for me at all unfortunately - when I read back the CPACR register after writing 0b11 to CP_10 and CP_11 they are still 0.
(I have no problems with an STM32F303K8 part or an STM32F411CE part - everything works as expected).

Would it be possible to add another (minimal) example project in the examples directory with this working?

djix123, 2021-05-28:
As a further update I received a few more BluePill-Plus / GD32F303 boards recently.

I built the GNU-template project that has been added to repository (thanks!) and enabled the FPU compiler settings. The new boards work perfectly fine - and the temperate and Vref is calculated using the FPU.
But, when I use my original boards the code hard faults with a NOCP fault per previous comments.
(same binary uploaded to the boards)

Looking at the processor markings the two sets of boards clearly have processors from different batches as the markings are different.

Liuyufanlyf, 2021-05-29:
I'm glad to hear that my contribution is helpful, keil is a powerful ide, but I believe open source toolchain is better.

Considering your situation, it's possible that some batches have a bug in fpu.

Have a nice day!

djix123, 2021-05-29:
Yes, thanks! Very helpful indeed! I've also attached some photos (not the best quality) of the MCU with working FPU and the MCU with a non-working FPU. Hope that's helpful.
[image]
[image]

djix123, 2022-05-22:
Eventually got around to replacing an MCU with a non-working FPU with a replacement part (GD32F303CG).
[Example project using the GigaDevice SPL](https://github.com/djix123/GD32F303CG_CMake)

hgl10, 2022-05-30:
我也遇到了类似的问题。我手里有两个BluePill Plus(GD32F303)，一个编号是JJ2042，一个是AC2137。编号是JJ2042的芯片，不能启用Single Precision，如果启用芯片会直接不工作，连闪烁LED灯都做不到。如果不启用就一切正常。但编号AC2137的板子都正常