WeActStudio/BluePill-Plus #7: not able to put it in boot mode

alinelena, 2021-04-20:
I am trying to follow the method described in here to boot..

https://www.aliexpress.com/item/4001116767437.html

both methods result in the same behaviour... the blue led is not flashing, only power led on. but on the host side.

```
[77157.651167] usb 1-4: new full-speed USB device number 109 using xhci_hcd
[77157.779302] usb 1-4: device descriptor read/64, error -71
[77158.036415] usb 1-4: New USB device found, idVendor=28e9, idProduct=0285, bcdDevice= 1.00
[77158.036427] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[77158.036433] usb 1-4: Product: GD32 USB MSC in FS Mode
[77158.036437] usb 1-4: Manufacturer: WeAct Studio
[77158.036440] usb 1-4: SerialNumber: GD32F30X-V3.0.0-3a4b5ec
[77168.303456] usb-storage 1-4:1.0: USB Mass Storage device detected
[77168.304040] scsi host0: usb-storage 1-4:1.0
[77169.327891] scsi 0:0:0:0: Direct-Access     GD32     Nand Flash       1.00 PQ: 0 ANSI: 0 CCS
[77169.328480] sd 0:0:0:0: Attached scsi generic sg0 type 0
[77169.329187] sd 0:0:0:0: [sda] 18 2048-byte logical blocks: (36.9 kB/36.0 KiB)
[77169.329474] sd 0:0:0:0: [sda] Write Protect is off
[77169.329480] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[77169.329811] sd 0:0:0:0: [sda] No Caching mode page found
[77169.329819] sd 0:0:0:0: [sda] Assuming drive cache: write through
[77169.352952]  sda:
[77169.354590] sd 0:0:0:0: [sda] Attached SCSI removable disk
[78402.287595] usb 1-4: reset full-speed USB device number 109 using xhci_hcd
[78402.415291] usb 1-4: device descriptor read/64, error -71
[78402.651349] usb 1-4: device descriptor read/64, error -71
[78402.887306] usb 1-4: reset full-speed USB device number 109 using xhci_hcd
[78403.015361] usb 1-4: device descriptor read/64, error -71
[78403.251344] usb 1-4: device descriptor read/64, error -71
[78403.487362] usb 1-4: reset full-speed USB device number 109 using xhci_hcd
[78403.487581] usb 1-4: Device not responding to setup address.
[78403.695519] usb 1-4: Device not responding to setup address.
[78403.907324] usb 1-4: device not accepting address 109, error -71
[78404.035330] usb 1-4: reset full-speed USB device number 109 using xhci_hcd
[78404.035499] usb 1-4: Device not responding to setup address.
[78404.243427] usb 1-4: Device not responding to setup address.
[78404.451322] usb 1-4: device not accepting address 109, error -71
[78404.451430] usb 1-4: USB disconnect, device number 109
[78404.639354] usb 1-4: new full-speed USB device number 110 using xhci_hcd
[78404.767298] usb 1-4: device descriptor read/64, error -71
[78405.003347] usb 1-4: device descriptor read/64, error -71
[78405.239318] usb 1-4: new full-speed USB device number 111 using xhci_hcd
[78405.367366] usb 1-4: device descriptor read/64, error -71
[78405.611371] usb 1-4: device descriptor read/64, error -71
[78405.719437] usb usb1-port4: attempt power cycle
[78406.375319] usb 1-4: new full-speed USB device number 112 using xhci_hcd
[78406.375502] usb 1-4: Device not responding to setup address.
[78406.583413] usb 1-4: Device not responding to setup address.
[78406.791348] usb 1-4: device not accepting address 112, error -71
[78406.919324] usb 1-4: new full-speed USB device number 113 using xhci_hcd
[78406.919482] usb 1-4: Device not responding to setup address.
[78407.127531] usb 1-4: Device not responding to setup address.
[78407.335323] usb 1-4: device not accepting address 113, error -71
[78407.335437] usb usb1-port4: unable to enumerate USB device
[78407.396235] FAT-fs (sda): unable to read boot sector to mark fs as dirty
```

closed 2021-06-26 by alinelena

Dans-Droids, 2021-11-27:
I have the same problem, did you fix the issue??

alinelena, 2021-11-27:
yap... training of the fingers... also having a9 a10 not connected to anything helped...are the selling them again?

Dans-Droids, 2021-11-27:
Haha, congrats.. I managed to solve it by erasing the entire chip and after abit of button mashing/erase, erase and erase again -  I was able to reupload the bootloader and upload a sketch..
I dont know if they are on sale again, mine is about 2 years old!

alinelena, 2021-11-29:
ok... i wonder if you got one with the "wrong" clock speed that is known to be a pain to reset. good you sorted it