WeActStudio/BluePill-Plus #8: arduino question

koendv, 2021-05-07:
I have a question. I can compile an arduino program if I treat the GD32F303 as a STM32F103.
But then I cannot use all ram and flash of the GD32F303.
Is there a boards.txt file for GD32F303 that allows to use the processor more fully?

closed 2021-05-12 by koendv

koendv, 2021-10-18:
I did not find a micropython port. 

There is support for Arduino: 
https://github.com/CommunityGD32Cores/ArduinoCore-GD32

and Huawei LiteOS also runs:
https://github.com/koendv/liteos-bluepillplus

koen

rohit-kumar-j, 2021-10-18:
@koendv Thanks for the quick reply!

I'm sure if this is the correct reason, but the lack of support seems to be the lower capacity of Ram (64Kb). Hence there is not much benefit for using an stm32 with micropython on this board if the support libraries, etc take up almost the entire space.